lhr-solar/Controls #11: Pedal Driver

JimothyGreene, 2020-06-02:
Driver software to interact with the accelerator and brake pedals. This should take input from the ADC and interpret it appropriately as a distance pressed for each pedal.

JimothyGreene, 2020-08-07:
What do you guys think about moving the `pedal_t` enum to this module and putting an `ADC_t `in the BSP_ADC module? Just to clarify which is which and keep the naming consistent. The `ADC_t` would be similar to the `pedal_t`, but it would consist of something like `ADC_0` and `ADC_1` so we keep the BSP_ADC module separated from the specific pieces of hardware that are connected to it. Let me know what you guys think.

ErickCortez98, 2020-08-07:
> What do you guys think about moving the `pedal_t` enum to this module and putting an `ADC_t `in the BSP_ADC module? Just to clarify which is which and keep the naming consistent. The `ADC_t` would [...]

Yeah! I think that would be one final detail that will make the ADC module independent of any specific hardware. I like it

closed 2020-08-29 by JimothyGreene