lhr-solar/Controls #169: [RTOS] Global Fault Handling

babusid, 2021-12-31:
# Description and Background
If we have an OS error that is not handled by the thread that it occurs in, we currently do not handle this properly the way BPS does. BPS's way of handling this is shown [here](https://github.com/lhr-solar/BPS/blob/c0aabfb1390afedae6c89bb5627b340984ad060a/Apps/Src/Tasks.c#L71) and [here](https://github.com/lhr-solar/BPS/blob/c0aabfb1390afedae6c89bb5627b340984ad060a/Apps/Src/Task_FaultState.c#L22). (Context provided below)

# BPS Solution
BPS has an application thread that pends on a global fault semaphore, and whenever a thread is unable to recover from an OS fault it attempts to post to that semaphore through the `assertOSError` function. This thread then runs the `EnterFaultState` function which accesses BSP (board layer code) functions to shut down the system safely. If the OS fault is such that posting to that fault semaphore is impossible, the `assertOSError` manually runs the `EnterFaultState` function. This `EnterFaultState` function is also bound to all of the board level fault interrupts (HardFault_Handler, BusFault_Handler, etc.)

# What needs to be done: 
- We're going to do the same thing BPS does
1. Create an `EnterFaultState` function that safely shuts down the motor and signals a controls fault. This function should ONLY rely on BSP level code, as we need to go directly to hardware level here.
2. Create a global fault semaphore 
3. Create a global fault Bitmap (a variable that has certain bits set depending on what kind of fault has occurred). The enterFaultState can rely upon this to do certain things in case of different fault types.
3. Create a `FaultHandler` task that pends on that semaphore
4. Create an `assertOSError` function that posts to that semaphore, and if that semaphore post fails, it then calls the EnterFaultState function directly. This function should accept an argument of the `OS_ERR` type, and if the argument provided is equivalent to `OS_ERR_NONE`, it shouldn't do anything. (This allows us to call this function in every thread without having to check that an error has actually occurred ahead of time)
5. Bind all Interrupt Fault Handlers to the EneterFaultState Function. In this case, the Fault_Bitmap should be set such that it is clear the fault has occurred due to an Interrupt Fault.

babusid, 2022-01-15:
I found an additional error case, [here](https://github.com/lhr-solar/Controls/blob/RTOS_SwitchRead/Apps/Src/ReadSwitches.c). On line 149. This wouldn't be an OS Error case but could be a different error type that we still need to assert.

IshDeshpa, 2022-01-22:
Additional error case found [here](https://github.com/lhr-solar/Controls/blob/master/Apps/Src/ReadCarCAN.c#L44). This kind of thing should be moved to the FaultHandler. Also, we are not handling a similar error case [here](https://github.com/lhr-solar/Controls/blob/master/Apps/Src/ReadTritium.c#L27). We should be doing something if ReadTritium is unsuccessful.

IshDeshpa, 2022-01-22:
We should make this a packed struct instead of a bitmap like BPS. A struct will have nicer names and take up the same memory + access speed as a bitmap.

IshDeshpa, 2022-01-22:
Not sure about whether we should integrate this [motor error stuff](https://github.com/lhr-solar/Controls/blob/master/Drivers/Src/MotorController.c#L91) into the global fault handler.

IshDeshpa, 2022-01-22:
> We should make this a packed struct instead of a bitmap like BPS. A struct will have nicer names and take up the same memory + access speed as a bitmap.

[Bit Field Declaration](https://www.tutorialspoint.com/cprogramming/c_bit_fields.htm)

IshDeshpa, 2022-01-22:
[link](url)

Ideally it should never reach this default condition but maybe some generic error here would be a good idea.

IshDeshpa, 2022-01-22:
Many of the [display functions](https://github.com/lhr-solar/Controls/blob/SendDisplay/Drivers/Src/Display.c#L87) return an ErrorStatus. Maybe we could have a generic display error that encompasses these? This isn't too high up on the priority list.

closed 2022-06-09 by ClarkPoon