lhr-solar/Controls #172: Task Suspension

Cam0Cow, 2022-01-22:
The current mechanism for suspending and resuming tasks is needlessly confusing, and doesn't allow other tasks to check if a task is suspended or not. This scheme should be replaced by a variable for each task, letting it know if it should be running or not. If not, the task should stop and pend on a semaphore. Otherwise, it'll run as normal. To resume a task, another tasks should set it's variable to the running state and then signal the task's semaphore.

Cam0Cow, 2022-01-22:
The variables should likely be in car state and might have to be in mutexes, since they'll be read from and written to from different tasks, potentially at the same time.

ClarkPoon, 2022-01-26:
Why are you using semaphores for this instead of just using `OSTaskResume()` and `OSTaskSuspend()`?

Cam0Cow, 2022-01-26:
> Why are you using semaphores for this instead of just using [`OSTaskResume()`](https://docs.silabs.com/micrium/latest/micrium-kernel-api/09-kernel-task-management-api#os-task-resume) and [...]

I didn't actually know that uCOS had these. I'll edit the issue to reflect that.

PR #173 (2022-02-19): New Task Suspension Code
Closes #172
closed 2022-02-19 by babusid