lhr-solar/Controls #176: Verify Minion Board GPIOExpander port friendliness

babusid, 2022-01-29:
One switch on the GPIO Expander (The reverse switch, GPIOB Pin 7) shares a port with the lights. We have to make absolutely sure that at no point does this switch get set as an output because that could theoretically cause a short that breaks the GPIO expander. The file that needs to be checked is the lights driver (this file sets up GPIOB for the lights).

babusid, 2022-02-14:
The lights and Switches initializations need to be checked.

IshDeshpa, 2022-04-02:
Checked with the KiCAD for the Minions Board; GPIOB Pin 7 is actually not connected to anything. The port for the reverse switch is connected to GPIOA Pin 2 and the only switch connected to GPIOB is the Hazard Switch on GPIOB Pin 6.

babusid, 2022-04-13:
@IshDeshpa is this done?

IshDeshpa, 2022-04-14:
@babusid Yeah it's gonna be closed in the Minions Driver pull request

closed 2022-04-16 by IshDeshpa