lhr-solar/Controls #293: # New BPS Fault routine + EDR Changes

babusid, 2023-02-04:
## Description
We need to ensure that we handle BPS faults properly. When BPS faults out, we will be powered off of the supplemental battery, but we need to ensure that we evacuate the driver. This will require a couple things.
1) We need to open both of our contactors. As of EDR, we control only the array bypass and motor bypass contactors. We need to open the array precharge contactor, and the motor bypass contactor.
2) We need to signal on the display that we have a BPS fault. This will require a new screen and icons on the display. This will also maybe need an update to the Fault handler to run this code.
3) We need to go into fault state. If BPS has faulted out, we should do nothing other than ensure the driver gets out of the car.

This will require modifications to the display firmware, the display driver, the fault handler, ReadCarCAN, and the CAN BPS. I think the first two are pretty self-explanatory, and @IshDeshpa we've talked about it, so I'll let you take the lead on that. The others, I will describe below. 

## Changes that I think are necessary
Firstly, we need to add a case in the ReadCarCan thread to handle the BPS Fault message. You can find the exact CAN ID in the CAN ID's sheet (@Cam0Cow can help with this, he should know where it is). If that message isn't supported by the CAN Driver, try adding an entry to the CANLUT and the ID enum, but if you don't feel comfortable message me. You should be able to add it yourself though. If you receive this message, you have to trigger a non-recoverable fault in the fault handler. This one doesn't require multiple trips, a single trip is enough to kill the system. This fault handler code will then do the display actions that are necessary. 

Secondly, we now know for sure that we don't control the array precharge contactor. We only need to deal with the array bypass contactor, and blind-wait for the precharge sequence to complete. Essentially, assume that the second you recieve a `charge_enable` message, BPS has closed the precharge contactor. Rework the ReadCarCAN code off of that assumption.

I also have some concerns regarding the safety of some of the variables used in the timer callback functions. I think we just need to go over it. I also see the `FAULT_BITMAP` being set, but I'm not sure where it's cleared? This is just something random that I spotted.

Finally, we need to setup low-level hardware filtering of CAN messages in the BSP layer. @IshDeshpa @millergage @KnockbackNemo If yall could work on this together this would be great. I want to make sure that we don't get flooded with CAN messages that we don't want to recieve. This will require dropping down to the BSP layer, which none of yall have done before. I'll be happy to help with this, but I think it's important that we get this done. I've also gone ahead and described a potential architecture for how to implement this (in pseudocode) below.

## Potential Architecture for CAN Filtering
1) Create a lookup table in the driver layer, consisting of all of the ID's that we actually want to read. This will be a simple array of `CAN_ID`'s. If you don't know which messages are critical, ask me, and we can talk about it. There should be an array for the main CAN bus, and an array for the Motor CAN bus.
2) Create a driver-layer API to take the correct lookup table, and pass a pointer to it, and the size to the BSP layer. There, it will be utilized by a BSP layer function to setup the hardware filters. Add a call to this function in the driver layer's init function.
3) In the BSP layer, create a function that takes a pointer to the lookup table, and the size of the lookup table. This function will then setup the hardware filters to only allow messages with the ID's in the lookup table to be received. It should also apply the filter only to the selected bus. This will require seome research into the Standard Peripheral Library's CAN filtering capabilities, but it is possible. Some helpful links:
    - [STD Periph documentation](http://www.disca.upv.es/aperles/arm_cortex_m3/curset/STM32F4xx_DSP_StdPeriph_Lib_V1.0.1/html/struct_c_a_n___filter_init_type_def.html#ac1368e531f4dd6ff08445ae664a9b96c)
    - [Example of someone using them](https://community.st.com/s/question/0D50X00009XkfSlSAJ/can-filters)
    - [Another example](https://schulz-m.github.io/2017/03/23/stm32-can-id-filter/)
    - [Another example](https://stackoverflow.com/questions/11724145/stm32f4discovery-can-filter-configuration)

## Suggested implementation approach
I suggest that @KnockbackNemo starts working with reworking the application layer (ReadCarCAN) with any changes that need to be made, while @millergage researches and starts on filters. @IshDeshpa, start with fault and display. 

This is a pretty long/big issue ticket, so if there's any confusion, message me!

babusid, 2023-07-11:
@IshDeshpa This issue ticket is probably out-of-date / needs to get broken up. Any particular reason you're keeping it around?

IshDeshpa, 2023-07-13:
This has been fully tested. Marking as closed.

closed 2023-07-13 by IshDeshpa