lhr-solar/Controls #335: Add Back EMF and Percent Power Output to Display

IshDeshpa, 2023-07-07:
Both can be calculated/obtained from our [motor controller](https://utexas.sharepoint.com/:b:/s/ENGR-LonghornRacing/EVb2-iGdYkdGkEMl44-lTvIBgV-iotwTGzEtiI2kKI31mA?e=Ot7kkp).

babusid, 2023-07-11:
@IshDeshpa What do you hope to gain (from a driver's perspective) by adding these? I would warn against making the driver consume too much information.

IshDeshpa, 2023-07-11:
The BackEMF would replace the accelerator slider at the bottom we have currently, as is present in most hybrid cars. Power percentage may not be completely necessary, but it could be put in the corner.

babusid, 2023-07-11:
You have limited screen real estate. It may in fact be more prudent to use any excess to size more important information larger, as you can't make assumptions about future placements, and it needs to be more versatile.

Cam0Cow, 2023-07-11:
Power is pretty important information, it'll let the driver know if they're being too heavy on the accelerator.