lhr-solar/Controls #365: Log diagnostic info in EEPROM

KnockbackNemo, 2023-10-14:
We want to make sure that our record of the car's FSM state (and other diagnostic info) is complete and accurate (in essence, no information is missed) so that we're able to trace the car's behavior (important for when the car fails and we want to find out what happened). Our car state data is currently sent to Data Aq over CANbus, but this makes it hard to guarantee consistent logging.

One solution is to log data ourselves using our EEPROM to make sure everything gets saved.

One could simply dedicate a region of flash to write to or create a file system to manage the logs.

*This issue ticket arose from attempting to ensure all FSM car state messages would be sent via CAN in the correct order without missing steps.*

NathanielDelgado, 2023-10-14:
1. Read this thing
2. Look up how to use Flash drivers
3. Ask questions
4. Start working 
@Arnav-Surjan

IshDeshpa, 2023-11-01:
This needs to be changed to be an odometer instead of the FSM state. Since flash memory has a limit to how many times we can write to it, and the FSM state changes too often, logging the number of miles driven is a safer bet while still offering utility.