lhr-solar/Controls #398: Fix SendTritium macro SENDTRITIUM_EXPOSE_VARS so that it can compile

KnockbackNemo, 2024-01-02:
**Describe the bug**
SendTritium currently has a macro SENDTRITIUM_EXPOSE_VARS that assists during debugging by exposing variables related to hardware input and hiding the code for reading them on hardware. However, SENDTRITIUM_EXPOSE_VARS currently doesn't work since #define-ing it creates compile errors from externing static variables and defining unused variables.

**To Reproduce**
Steps to reproduce the behavior:
1. Define SENDTRITIUM_EXPOSE_VARS in SendTritium on the master branch
2. Compile leader (main or a test)

**Expected behavior**
Defining SENDTRITIUM_EXPOSE_VARS should disable CAN messages and reading inputs and instead allow you to set/inspect variables in a test file by externing them in SendTritium.h.

NathanielDelgado, 2024-01-13:
Seems to be fine in dev/code_cleanup

KnockbackNemo, 2024-01-16:
Cool, I guess the changes there fix it! I'll link this to the code cleanup PR.

closed 2024-01-20 by NathanielDelgado